HestiaPi/hestia-touch-openhab #10: Unify generic.rules with hvac.rules

gulliverrr, 2019-11-02:


rkoshak, 2020-02-11:
This issue can be closed as a duplicate of #26, or #26 closed as a duplicate of this. I have the code done and am systematically testing it. I didn't get as much of a reduction in number of lines of code over all, unless you count all the lines from both .rules files. I'm currently sitting at around 850 lines of code in one file that handles both US and EU types. But those 850 includes lots of comments and logging too. 

What do you want to call this unified .rules file? From looking at the scripts we can't use default.rules. I'm using "hestiapi-one.rules" at the moment but want to get the name right before I check stuff in.

gulliverrr, 2020-02-11:
If all rules will be included in one file, then it should be named default.rules but:

1. The generic.rules and hvac.rules should be deleted
2. Store in the appropriate location (/etc/openhab2/rules/default.rules) so that others find it easily too
3. The [startup script](pi/scripts/openhabloader.sh) should be changed to handle these changes

I'm happy to do 1 and 3.

rkoshak, 2020-02-11:
I was hesitant to use default.rules because both kiosk-xinit.sh and openhabloader.sh will move /etc/openhab2/rules/default.rules to /home/pi/scripts/default.rules and then later copies it back, or if there is no default copies the "stock" .rules file back.

Based on the little bit of looking I've done with the upgrade script, if we call it default.rules, on that first reboot, the newly downloaded from github .rules file will be overwritten with the old file /etc/openhab2/rules/default.rules.

Also, I like that there is a copy of the "stock" .rules file that remains in the scripts folder which makes it relatively easy for users hacking away on their Rules to go back to stock. Because of the moving of the "live" default.rules file back to scripts we would lose that. 

Ultimately, I would like to see this hiding of the .rules file stuff implemented similar to how openHABian does it, by editing the openhab2.service file. That way every time you start/restart openHAB, whether it's a reboot of the RPi or  `sudo systemctl restart openhab2` the .rules files will be hidden for the appropriate amount of time. As currently implemented, they only get hidden on a full system reboot.

1. will be done in my PR.
2. will be done as well, though I'm not using default.rules. Based on the outcome of this conversation I'll use the appropriate name.
3. I've already made the changes to openhabloader.sh *and* kiosk-xinit.sh to use the name above. But like I said, I'm concerned about upgrades and losing either the user's original .rules file or having the user's .rules file overwrite the newly released one. Neither sounds like a good idea to me.

gulliverrr, 2020-02-11:
Once default.rules is added to the next release of GitHub (1.2) it will be the one to be downloaded by the update script.

gulliverrr, 2020-02-14:
Solved by #28

gulliverrr, 2020-02-14:
Solved by #28

closed 2020-02-14 by gulliverrr