HestiaPi/hestia-touch-openhab #11: Enable phone notifications from UI

gulliverrr, 2019-11-07:
Allow user to enable/disable notification (groups) from the phone app or web UI.

rkoshak, 2020-02-25:
I assume this is using myopenhab.org? If so this should be simple enough.

1. Install the Cloud Connector as part of the base install.
2. Have the user sign up for an account and inform them how to link their HestiaPi to the server (NOTE: if they already have an openHAB instance there will be problems as myopenhab.org can only support one OH instance per account.)
3. Create a Switch Item to turn notifications ON and OFF.
4. Create a String Item that get's commanded with alerts.
5. Create a Rule that gets triggered when the String Item is commanded. The Rule checks the Switch and if it's ON it sends a broadcast notification containing the state of the String Item. 

Once can even set up alert levels for these notifications. Instead of a Switch we can use a String Item and use a Selection or Mappings on the sitemap to choose what level to alert to. 

But the question is, what should be alerted? What events warrant an alert and what level? I saw a couple of commented out email alerting lines in the code but surely there is a desire for more than just those events. 

Personally, I can't think of when I'd want to get alerts from the thermostat so I am not sure the best way to implement the actual alerts to send. The rest I've already implemented in my main openHAB instance and even wrote a design pattern tutorial based on it. 

What might be useful though is a monthly email with a report showing how much each device was used over the past month. But the only way for the user to enter their email account information through a UI would be if we upgrade to 2.5 and use the new Mail binding. Then they can use PaperUI to set up the email or we can provide a UI that creates the Thing through the REST APi.

gulliverrr, 2020-02-25:
I see upgrade to 2.5 climbing near the top of the list here :)
I didn't know of `sendBroadcastNotification(message)` apart from `sendNotification(emailAddress, message)` so that is way much easier like this.
Use cases that come to my mind:
- Heating/Cooling/Hot water started
- Heating/Cooling/Hot water temperature reached
- Setpoint changes: For houses monitored (elderly, babysitters, cleaners)
- In case of remote sensors:
  - Temp/Humidity Sensor goes offline
  - Flood sensor detects water
  - Presence sensor detect presence during 'away' hours
  - Alarm system gets triggered

Of course some are for functionality that is not enabled by default but it would be great to have the process in place and allow the user to include it or not to their custom additions.