HestiaPi/hestia-touch-openhab #17: Suppress "Asynchronous processing not supported" warning in log

rkoshak, 2020-01-21:
Raise the level of the org.eclipse.jetty.server.HttpChannel logger to ERROR to suppress these warnings when BasicUI is open.

Add the following to the bottom of `/var/lib/openhab2/etc/org.ops4j.pax.logging.cfg`

```
# HestiaPi

# Filter out Asynchronous processing not supported warnings
log4j2.logger.org_eclipse_jetty_server_httpchannel.level = ERROR
log4j2.logger.org_eclipse_jetty_server_httpchannel.name = org.eclipse.jetty.server.HttpChannel
```

You may need to restart OH for the changes to take effect. 

Alternatively open the Karaf console:

    ssh -p 8101 openhab:localhost

Default password is habopen

Issue the command:

    log:set ERROR org.eclipse.jetty.server.HttpChannel

This will immediately take effect and add those two lines to the logging config file.

gulliverrr, 2020-02-14:
@rkoshak Is there a (preferred) way for the above to be implemented persistently in this repo or should it be added with a startup script?

rkoshak, 2020-02-17:
Ideally, I'd like to see /var/lib/openhab2 included in this repo. There are additional changes I want to make to the etc files (e.g. remove the LSP service) and if we migrate to JSONDB configs we will need the JSONDB files. We would ignore the cache and tmp folders of course, and perhaps even the persistence and mqtt folders and definitely the backups folders. 

If we configuration control these folders too, that will mean that config changes like the above can be captured and preserved. And if I can get all the configs migrated to JSONDB there will be no more need to wait for everything to come up before parsing .rules files, the system will start up *much* faster, and users will be able to modify/customize everything through PaperUI. 

I do think migrating to JSONDB for everything is the only thing that will drop the reboot time significantly at this point. 

If you look in the backup zip file produced by openhab-cli, you will see it includes the above folders too.

But it does impose a little bit of extra pain on us as developers because PaperUI isn't really great for Rules (OH 3 will have a better and fully functional Rules editing capability in the replacement to PaperUI). But on the plus side it means we can use a "real" language, in this case JavaScript. 

To start though, I'd just add /var/lib/openhab2/etc and update the install/update scripts to handle that folder. Then we can gradually add more as things progress.

rkoshak, 2020-04-22:
This Issue can be closed. It's been implemented in PR #38.

closed 2020-04-22 by rkoshak