HestiaPi/hestia-touch-openhab #31: Humidity Control not updating LCD or Pin

rkoshak, 2020-02-17:
For EU/Generic systems, humidity control and hot water have the following same issue:

Web UI shows correctly the state as ON
LCD UI shows incorrectly the state as OFF (icon stays white)
Pin and relay is OFF

gulliverrr, 2020-02-18:
Humidity was a typo. 75d87e6 fixes it. Hot water is not depending on a sensor value change so pin state still not getting updated on boot.

rkoshak, 2020-02-18:
Hmmmm, that's off because I made that fix already.  But if I understand correctly, that typo fixed the humidity, so only hot water is failing to update on boot but otherwise working correctly?

Can you post events.log between the times where openhab.log shows "Initializing settings" and "Initializing settings" plus a second (just in case).

I'm looking at the latest in ONE and some of my changes don't seem to be there. I'm wondering if I failed to do the merge correctly for the Things PR. In this case, the line that sets `initializing` to false needs to be moved before the for loop that turns off boost modes and there is a missing additional for loop that re sendCommands the Modes to force the devices to operate based on the current state instead of waiting for a sensor update. 

I'll make a PR to restore those changes and fix anything else that seems to have been lost somehow.

rkoshak, 2020-02-18:
OK, everything should be working now. At the end of initializing I just call the lambdas directly instead of trying to kick off the Mode rule or the sensor reading changed rule. This will work for everything and appears to work based on my tests.

gulliverrr, 2020-02-18:
> Can you post events.log between the times where openhab.log shows "Initializing settings" and "Initializing settings" plus a second (just in case).

events.log:
```
2020-02-18 12:10:42.641 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.6
2020-02-18 12:10:42.659 [vent.ItemStateChangedEvent] - MyHumiProxy changed from NULL to 36.6
2020-02-18 12:10:42.684 [ome.event.ItemCommandEvent] - Item 'SystemType' received command EU
2020-02-18 12:10:42.692 [nt.ItemStatePredictedEvent] - SystemType predicted to become EU
2020-02-18 12:10:42.826 [ome.event.ItemCommandEvent] - Item 'TempUnit' received command F
2020-02-18 12:10:42.834 [nt.ItemStatePredictedEvent] - TempUnit predicted to become F
2020-02-18 12:10:44.642 [vent.ItemStateChangedEvent] - MyHumi changed from 36.6 to 36.7
2020-02-18 12:10:54.821 [vent.ItemStateChangedEvent] - MyHumi changed from 36.7 to 36.8
2020-02-18 12:10:59.312 [ome.event.ItemCommandEvent] - Item 'MyHumiProxy' received command 36.8
2020-02-18 12:10:59.326 [ome.event.ItemCommandEvent] - Item 'HumidityType' received command Dehumidify
2020-02-18 12:10:59.349 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.8
2020-02-18 12:10:59.354 [nt.ItemStatePredictedEvent] - HumidityType predicted to become Dehumidify
2020-02-18 12:10:59.367 [vent.ItemStateChangedEvent] - MyHumiProxy changed from 36.6 to 36.8
2020-02-18 12:10:59.373 [ome.event.ItemCommandEvent] - Item 'MyHumiProxy' received command 36.8
2020-02-18 12:10:59.380 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.8
2020-02-18 12:10:59.408 [ome.event.ItemCommandEvent] - Item 'MainSwitch' received command ON
2020-02-18 12:10:59.656 [ome.event.ItemCommandEvent] - Item 'HeatingMode' received command ON
2020-02-18 12:10:59.693 [ome.event.ItemCommandEvent] - Item 'HotWaterMode' received command ON
2020-02-18 12:10:59.955 [ome.event.ItemCommandEvent] - Item 'HumidityMode' received command ON
2020-02-18 12:10:59.966 [ome.event.ItemCommandEvent] - Item 'Heating2' received command OFF
2020-02-18 12:10:59.981 [ome.event.ItemCommandEvent] - Item 'HumidityType' received command Dehumidify
2020-02-18 12:10:59.989 [nt.ItemStatePredictedEvent] - HumidityType predicted to become Dehumidify
2020-02-18 12:11:00.028 [ome.event.ItemCommandEvent] - Item 'TempSetpoint' received command 90
2020-02-18 12:11:00.060 [ome.event.ItemCommandEvent] - Item 'TempSetpointF' received command 90
2020-02-18 12:11:00.090 [ome.event.ItemCommandEvent] - Item 'HumiSetpoint' received command 25
2020-02-18 12:11:00.107 [ome.event.ItemCommandEvent] - Item 'HeatingBoostTime' received command 10
2020-02-18 12:11:00.129 [ome.event.ItemCommandEvent] - Item 'HotWaterBoostTime' received command 10
2020-02-18 12:11:00.184 [ome.event.ItemCommandEvent] - Item 'HumiBoostTime' received command 10
2020-02-18 12:11:00.191 [ome.event.ItemCommandEvent] - Item 'PreviousTempReading' received command 73
2020-02-18 12:11:00.215 [ome.event.ItemCommandEvent] - Item 'PreviousHumiReading' received command 37.0
2020-02-18 12:11:00.223 [ome.event.ItemCommandEvent] - Item 'Heating2Time' received command 3
2020-02-18 12:11:00.236 [ome.event.ItemCommandEvent] - Item 'Heating2Delta' received command 2
2020-02-18 12:11:00.422 [ome.event.ItemCommandEvent] - Item 'Network_WLAN_IP' received command 10.10.10.52
2020-02-18 12:11:00.427 [nt.ItemStatePredictedEvent] - Network_WLAN_IP predicted to become 10.10.10.52
2020-02-18 12:11:00.480 [ome.event.ItemCommandEvent] - Item 'Network_SSID' received command BOB
2020-02-18 12:11:00.497 [nt.ItemStatePredictedEvent] - Network_SSID predicted to become xxx
2020-02-18 12:11:00.598 [ome.event.ItemCommandEvent] - Item 'Network_WLAN_INFO' received command 92
2020-02-18 12:11:00.625 [nt.ItemStatePredictedEvent] - Network_WLAN_INFO predicted to become 92
2020-02-18 12:11:00.630 [vent.ItemStateChangedEvent] - Network_WLAN_INFO changed from 86 to 92
2020-02-18 12:11:00.667 [ome.event.ItemCommandEvent] - Item 'Network_WLAN_MAC' received command b8:27:eb:xx:xx:43
2020-02-18 12:11:00.684 [nt.ItemStatePredictedEvent] - Network_WLAN_MAC predicted to become b8:27:eb:xx:xx:43
2020-02-18 12:11:00.697 [vent.ItemStateChangedEvent] - Network_WLAN_MAC changed from NULL to b8:27:eb:xx:xx:43
2020-02-18 12:11:00.773 [ome.event.ItemCommandEvent] - Item 'System_CPU_TEMP' received command 122 °F
2020-02-18 12:11:00.788 [nt.ItemStatePredictedEvent] - System_CPU_TEMP predicted to become 122 °F
2020-02-18 12:11:01.859 [ome.event.ItemCommandEvent] - Item 'System_CPU_LOAD' received command 37
2020-02-18 12:11:01.866 [nt.ItemStatePredictedEvent] - System_CPU_LOAD predicted to become 37
2020-02-18 12:11:01.870 [vent.ItemStateChangedEvent] - System_CPU_LOAD changed from 49 to 37
2020-02-18 12:11:04.991 [vent.ItemStateChangedEvent] - MyHumi changed from 36.8 to 36.9
2020-02-18 12:11:19.044 [ome.event.ItemCommandEvent] - Item 'SystemType' received command EU
2020-02-18 12:11:19.049 [ome.event.ItemCommandEvent] - Item 'MyHumiProxy' received command 36.9
2020-02-18 12:11:19.060 [nt.ItemStatePredictedEvent] - SystemType predicted to become EU
2020-02-18 12:11:19.069 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.9
2020-02-18 12:11:19.082 [ome.event.ItemCommandEvent] - Item 'TempUnit' received command F
2020-02-18 12:11:19.085 [ome.event.ItemCommandEvent] - Item 'HumidityType' received command Dehumidify
2020-02-18 12:11:19.092 [vent.ItemStateChangedEvent] - MyHumiProxy changed from 36.8 to 36.9
[... cut ...]
2020-02-18 12:11:21.037 [INFO ] [marthome.model.script.initialization] - Sending 36.9 to MyHumiProxy
2020-02-18 12:11:21.063 [INFO ] [marthome.model.script.initialization] - Sending 0 to PreviousHumiReading
2020-02-18 12:11:21.080 [INFO ] [smarthome.model.script.temp-setpoint] - Turning ON the heater
2020-02-18 12:11:21.085 [INFO ] [marthome.model.script.initialization] - Done initializing settings
2020-02-18 12:11:21.136 [INFO ] [smarthome.model.script.humi-setpoint] - Turning ON Dehumidify
```

Testing your commit now. Will update.

gulliverrr, 2020-02-18:
> Can you post events.log between the times where openhab.log shows "Initializing settings" and "Initializing settings" plus a second (just in case).

events.log:
```
2020-02-18 12:10:42.641 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.6
2020-02-18 12:10:42.659 [vent.ItemStateChangedEvent] - MyHumiProxy changed from NULL to 36.6
2020-02-18 12:10:42.684 [ome.event.ItemCommandEvent] - Item 'SystemType' received command EU
2020-02-18 12:10:42.692 [nt.ItemStatePredictedEvent] - SystemType predicted to become EU
2020-02-18 12:10:42.826 [ome.event.ItemCommandEvent] - Item 'TempUnit' received command F
2020-02-18 12:10:42.834 [nt.ItemStatePredictedEvent] - TempUnit predicted to become F
2020-02-18 12:10:44.642 [vent.ItemStateChangedEvent] - MyHumi changed from 36.6 to 36.7
2020-02-18 12:10:54.821 [vent.ItemStateChangedEvent] - MyHumi changed from 36.7 to 36.8
2020-02-18 12:10:59.312 [ome.event.ItemCommandEvent] - Item 'MyHumiProxy' received command 36.8
2020-02-18 12:10:59.326 [ome.event.ItemCommandEvent] - Item 'HumidityType' received command Dehumidify
2020-02-18 12:10:59.349 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.8
2020-02-18 12:10:59.354 [nt.ItemStatePredictedEvent] - HumidityType predicted to become Dehumidify
2020-02-18 12:10:59.367 [vent.ItemStateChangedEvent] - MyHumiProxy changed from 36.6 to 36.8
2020-02-18 12:10:59.373 [ome.event.ItemCommandEvent] - Item 'MyHumiProxy' received command 36.8
2020-02-18 12:10:59.380 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.8
2020-02-18 12:10:59.408 [ome.event.ItemCommandEvent] - Item 'MainSwitch' received command ON
2020-02-18 12:10:59.656 [ome.event.ItemCommandEvent] - Item 'HeatingMode' received command ON
2020-02-18 12:10:59.693 [ome.event.ItemCommandEvent] - Item 'HotWaterMode' received command ON
2020-02-18 12:10:59.955 [ome.event.ItemCommandEvent] - Item 'HumidityMode' received command ON
2020-02-18 12:10:59.966 [ome.event.ItemCommandEvent] - Item 'Heating2' received command OFF
2020-02-18 12:10:59.981 [ome.event.ItemCommandEvent] - Item 'HumidityType' received command Dehumidify
2020-02-18 12:10:59.989 [nt.ItemStatePredictedEvent] - HumidityType predicted to become Dehumidify
2020-02-18 12:11:00.028 [ome.event.ItemCommandEvent] - Item 'TempSetpoint' received command 90
2020-02-18 12:11:00.060 [ome.event.ItemCommandEvent] - Item 'TempSetpointF' received command 90
2020-02-18 12:11:00.090 [ome.event.ItemCommandEvent] - Item 'HumiSetpoint' received command 25
2020-02-18 12:11:00.107 [ome.event.ItemCommandEvent] - Item 'HeatingBoostTime' received command 10
2020-02-18 12:11:00.129 [ome.event.ItemCommandEvent] - Item 'HotWaterBoostTime' received command 10
2020-02-18 12:11:00.184 [ome.event.ItemCommandEvent] - Item 'HumiBoostTime' received command 10
2020-02-18 12:11:00.191 [ome.event.ItemCommandEvent] - Item 'PreviousTempReading' received command 73
2020-02-18 12:11:00.215 [ome.event.ItemCommandEvent] - Item 'PreviousHumiReading' received command 37.0
2020-02-18 12:11:00.223 [ome.event.ItemCommandEvent] - Item 'Heating2Time' received command 3
2020-02-18 12:11:00.236 [ome.event.ItemCommandEvent] - Item 'Heating2Delta' received command 2
2020-02-18 12:11:00.422 [ome.event.ItemCommandEvent] - Item 'Network_WLAN_IP' received command 10.xx.xx.xx
2020-02-18 12:11:00.427 [nt.ItemStatePredictedEvent] - Network_WLAN_IP predicted to become 10.xx.xx.xx
2020-02-18 12:11:00.480 [ome.event.ItemCommandEvent] - Item 'Network_SSID' received command xxx
2020-02-18 12:11:00.497 [nt.ItemStatePredictedEvent] - Network_SSID predicted to become xxx
2020-02-18 12:11:00.598 [ome.event.ItemCommandEvent] - Item 'Network_WLAN_INFO' received command 92
2020-02-18 12:11:00.625 [nt.ItemStatePredictedEvent] - Network_WLAN_INFO predicted to become 92
2020-02-18 12:11:00.630 [vent.ItemStateChangedEvent] - Network_WLAN_INFO changed from 86 to 92
2020-02-18 12:11:00.667 [ome.event.ItemCommandEvent] - Item 'Network_WLAN_MAC' received command b8:27:eb:xx:xx:43
2020-02-18 12:11:00.684 [nt.ItemStatePredictedEvent] - Network_WLAN_MAC predicted to become b8:27:eb:xx:xx:43
2020-02-18 12:11:00.697 [vent.ItemStateChangedEvent] - Network_WLAN_MAC changed from NULL to b8:27:eb:xx:xx:43
2020-02-18 12:11:00.773 [ome.event.ItemCommandEvent] - Item 'System_CPU_TEMP' received command 122 °F
2020-02-18 12:11:00.788 [nt.ItemStatePredictedEvent] - System_CPU_TEMP predicted to become 122 °F
2020-02-18 12:11:01.859 [ome.event.ItemCommandEvent] - Item 'System_CPU_LOAD' received command 37
2020-02-18 12:11:01.866 [nt.ItemStatePredictedEvent] - System_CPU_LOAD predicted to become 37
2020-02-18 12:11:01.870 [vent.ItemStateChangedEvent] - System_CPU_LOAD changed from 49 to 37
2020-02-18 12:11:04.991 [vent.ItemStateChangedEvent] - MyHumi changed from 36.8 to 36.9
2020-02-18 12:11:19.044 [ome.event.ItemCommandEvent] - Item 'SystemType' received command EU
2020-02-18 12:11:19.049 [ome.event.ItemCommandEvent] - Item 'MyHumiProxy' received command 36.9
2020-02-18 12:11:19.060 [nt.ItemStatePredictedEvent] - SystemType predicted to become EU
2020-02-18 12:11:19.069 [nt.ItemStatePredictedEvent] - MyHumiProxy predicted to become 36.9
2020-02-18 12:11:19.082 [ome.event.ItemCommandEvent] - Item 'TempUnit' received command F
2020-02-18 12:11:19.085 [ome.event.ItemCommandEvent] - Item 'HumidityType' received command Dehumidify
2020-02-18 12:11:19.092 [vent.ItemStateChangedEvent] - MyHumiProxy changed from 36.8 to 36.9
[... cut ...]
2020-02-18 12:11:21.037 [INFO ] [marthome.model.script.initialization] - Sending 36.9 to MyHumiProxy
2020-02-18 12:11:21.063 [INFO ] [marthome.model.script.initialization] - Sending 0 to PreviousHumiReading
2020-02-18 12:11:21.080 [INFO ] [smarthome.model.script.temp-setpoint] - Turning ON the heater
2020-02-18 12:11:21.085 [INFO ] [marthome.model.script.initialization] - Done initializing settings
2020-02-18 12:11:21.136 [INFO ] [smarthome.model.script.humi-setpoint] - Turning ON Dehumidify
```

Testing your commit now. Will update.

closed 2020-02-18 by gulliverrr

gulliverrr, 2020-02-18:
@rkoshak 'sometimes' I get this error on boot:
`2020-02-19 00:17:36.388 [ERROR] [ntime.internal.engine.RuleEngineImpl] - Error during the execution of startup rule 'Initialization': Could not cast NULL to java.lang.Number; line 53, column 49, length 27`
without updating the `MyTempProxy`
I work on a Pi3 which may call of finish some rules faster than a Pi Zero...
After some time, everything clears and is back to normal. Not sure if it's a sensor updating issue.

rkoshak, 2020-02-18:
Line 53 is where I calculate the difference between the current temp and target temp. And indeed that line would throw an error if MyTemp or TempSetpoint are NULL. The timing is such that I'm not surprised MyTemp would be NULL on an RPi3. 

But lines 50-52 is supposed to check for that case and return with a meaningful error (". That's the odd part. DOH! I forgot the `return;`. Can you confirm that you are seeing 

    "TempSetpoint == 21 and MyTempProxy == NULL"

in openHAB log right before that error message?

I also noticed that I'm missing a similar check for cooling and humidity.

I'm working on updating boost mode. I'll include the fixes in that PR. In the mean time, insert a return; after the logError on line 50 and this error should go away.

I would only expect to see that error on an RPi 3. On the zero, there is no chance that the Initialization rule will get to the end before the sensors start reporting which would explain why I've not see this.

gulliverrr, 2020-02-18:
> Can you confirm that you are seeing
> 
> ```
> "TempSetpoint == 21 and MyTempProxy == NULL"
> ```
> 
> in openHAB log right before that error message?

No, I don't.
```
2020-02-19 00:17:34.246 [INFO ] [marthome.model.script.initialization] - Initializing settings
2020-02-19 00:17:36.388 [ERROR] [ntime.internal.engine.RuleEngineImpl] - Error during the execution of startup rule 'Initialization': Could not cast NULL to java.lang.Number; line 53, column 49, length 27
2020-02-19 00:18:11.068 [INFO ] [smarthome.model.script.temp-setpoint] - Turning ON the heater
2020-02-19 00:18:11.133 [INFO ] [smarthome.model.script.temp-setpoint] - Scheduling second stage heating to start in 0 minutes.
2020-02-19 00:18:11.423 [INFO ] [smarthome.model.script.temp-setpoint] - Turning on second stage heating
```

Does `UnDefType` catch `NULL` too?

rkoshak, 2020-02-18:
Yes, UnDefType should catch both NULL and UNDEF. I'll change to do the longer and more explicit check for both NULL and UNDEF. Though that should work with instanceof UnDefType...

rkoshak, 2020-02-19:
PR #33 includes the fixes for this.

gulliverrr, 2020-02-19:
Testing and reporting back...

gulliverrr, 2020-02-19:
Typo 48f74e7

PR #32 (2020-02-18): Fix for hotwater not restarting on reboot
Fixes #31.

Because the Mode Items do not change when recommanding them the Mode Rule never executes to kick off the changes to the devices. Previously I kicked things off by changing the sensor readings but HotWater is not based on a sensor reading. So I change to just call the lambdas directly for all the devices. Therefore, at the end of initializing if the Mode is ON, the device will turn ON.