HestiaPi/hestia-touch-openhab #37: Move configs to JSONDB

rkoshak, 2020-03-23:
I think I've made it to the point now where I can check in my second rewrite of the rules and other configs. At a high level here is what I've done:

- Moved all the Things from .things files to JSONDB.

- Created some new Exec Things to replace executeCommandLine calls in the Rules. Startup is a little faster when we let the Exec binding handle these calls instead of Rules.

- Moved all the Items from .items files except for the five with GPIO binding configs. This improves startup by greatly reducing how many lines of .items files need to be read and parsed.  This also allows users to see and more easily change/customize Items (e.g. translating to their native languages) through PaperUI instead of needing to ssh and modify files.

- Moved all the Rules to JSONDB written in JavaScript. I chose JavaScript over Python as I've noticed on my own machines that Python, like Rules DSL, takes a good deal of time to load and parse. JavaScript loads *very* fast but there is a slight delay (one or two seconds) the first time a Rule executes. Because the JSONDB rules work differently in many ways, there were some changes to the structure of the Rules. But I also was able to add some additional error checking (e.g. if you change the FanMode to OFF while Heating or Cooling is ON it will prevent it and restore the FanMode to Auto, it will not accept SystemType or HumidityType or TempUnit values that are not valid, restoring them to their previous value, etc.).

- Removed the Language Server Provider. It's not used, never going to be used and it takes up both memory and CPU during boot time. 

- Changed the kiosk script to wait for the system load to drop to 2.0 instead of 0.5. I found it was spending multiple minutes to drop to 0.5 and sometimes it never got that low because all the rules will have loaded and started running which wasn't the case before.

- There is still a default.rules file which gets copied to the rules folder when the load drops. This is done for two reasons. 1. the JSONDB rules do not yet support a System started trigger and 2. we don't want to run the initialization until the LCD is ready to receive the commands generated by initialization so the LCD matches the restored states.

- Lots more logging adding that lets us see what is going on.

As a result of the above I have managed to whittle down the boot time from the 25-30 minutes to almost exactly 10 from the white screen on the LCD until it's fully functional. I've made that 500 second countdown on the loading screen true. ;-D and sometimes even beat it.

Advantages:
- Users can see all the relevant configs in PaperUI. I know that PaperUI is not great, but for users who are not Linux experienced it's a godsend. This will let all levels of users see and modify all the relevant settings from the browser. For example, they can change the labels for Items to their native language. 

- They can add or change which topic is subscribed to for the sensor readings (change source of sensor data to a remote sensor) from the UI. 

- They can see the Rules and see them running (the status icon next to each Rule turns green when a Rule is running). 

- They can disable/enable/modify rules within the browser. For example, if they want to change the Heating behavior, they can disable the default Rule and create their own without changing the default one. If something goes wrong, they can just reenable the original one. 

- The JSONDB Rules have three clauses, one of which is the "but only if..." or conditional clause. This lets you define what conditions need to be true before a Rule can execute. I've added lots of additional error checking to these clauses. 

- Doing things like letting users configure email and create alerts through PaperUI will be possible.

Disadvantages:

- PaperUI Sucks! I admit it. It's clunky, hard to use, and requires lots of repetitive actions. I actually didn't use it much at all doing the above and instead create and/or modified the raw JSON and used the REST API instead. Hopefully most of the users will not be making massive changes so this won't be that big of a deal. But it's certainly not fun to use if you have to do a lot of creating.

- Your job is going to be harder as the JSONDB is not very human friendly to read and review. If needed we might need to make a script to extract the Rules and make them easier to read. The problem is the actual JavaScript code ends up with all the new lines replaced with URL encoding. As a result the code all runs together. But the good news is no block of code is more than a dozen or so lines of code, most are under 10.  Separating the error checking from the logic makes the code very short.

To help you review and help users understand the Rules so they can modify them, I've been working on some diagrams showing how the main Rules interact with each other. Here's an example:

[image]

Here are the things I don't know how to best handle as they have to deal with the install image:

- The JSONDB and other configs configs are stored in /var/lib/openhab2. I can and will check those in to this repo under var/lib/openhab2 but don't know everything that needs to change to make sure they get copied to the right place during an upgrade and the initial install when you create the SD image.

- I think this is OK as is, but you cannot upgrade these files while OH is running. I think the restore script already reboots, does the copying of files around and then reboots again. But on the first reboot, before copying the files, the script needs to kill OH before copying the files.

- If we can delay the checking of the CPU load in the kiosk script a bit, I think we can shave another 10-30 seconds off of the boot time. But I don't know the implications of that. What I really want is the while loop that checks for the system load before switching the LCD UI to the working UI to look for a load of 2.5. But I can't use anything over 2.0 because while OH is first starting the load doesn't jump above 2.5 fast enough. So if we can delay that checking until after it first goes above 2.5 we don't have to wait as long for the system load to drop before switching over. Even at around a system load of 3 OH is functional and ready to go and in fact the rules that process the sensor readings will already be running. 

Before I dump a whole lotta changes on you I wanted to open up a discussion and find out how you want me to proceed.

gulliverrr, 2020-03-24:
Wow.... That is very well explained! 
About your last 3 points...
* If you pass me the code (I believe a PR would be fine) I can handle the creation of the new image.
Could you have a look on the [script](https://github.com/HestiaPi/hestia-touch-openhab/blob/ONE/home/pi/scripts/packitupandgo.sh) that clears anything that needs to be cleared before the image is created and add maybe more new entries needed?
* I will amend the upgrade script. Right now it only copies the files off GitHub without stopping it and reboots. I'm not sure if I can have a script keep running after OH is killed. I'll find a workaround.
* I can make the load checking as advanced as you describe it, no problem. Need to make it work for people running it for testing/development on a Pi 3 or anything faster than a Zero (which practically means any other Pi)

rkoshak, 2020-03-24:
Add to the script to delete `/var/lib/openhab2/persistence/mapdb/*`. This will give the users a clean slate and make sure everything is initialized to their default values (which will be defined in `/etc/openhab2/automation/lib/hestia/defaults.js`) on the first boot.  From that point onward, their settings will be saved and restored but we need to boot strap the settings somehow. Alternatively, one can command all the setting Items to their defaults and leave the mapdb files alone. Then restoreOnStartup will initialize the Items to their defaults and then perhaps we can reduce the initialization rule.

The deletion of the rrd4j files is probably unnecessary as rrd4j is not even installed, unless you have a modified version of OH you build the image from that includes rrd4j.

Add deletion of `/var/lib/openhab2/jsondb/backup/*`. These are automatically created backups of the JSONDB which are made every time you make a change to any of the configs through PaperUI or the REST API (a handy feature for the users too). 

I believe this will give everyone a nice clean environment on the first boot.

For the backup, you might need to do something like you do with the restore script where you set a file, reboot, the boot script looks for that file and instead of loading the kiosk and all that stuff instead kills OH and copies down the new configs. Making a backup of the existing configs first would not be a bad idea. One gotcha with upgrades from v1.1 to whatever number this becomes will be the need to upgrade to OH 2.5.3 too. I'm not sure the best way to handle that. Maybe releasing a v1.1.1 or the like with changes to the upgrade script to apt upgrade OH and then an upgrade to v1.2.

On that note, when you install openHAB using apt on the "gold image", use `apt-get install openhab=2.5.3-1 -V` which will fix OH to that specific version and prevent it from being upgraded even when users run `apt dist-upgrade` or `apt upgrade`. Using the same approach for Mosquitto and any other service you want to keep at a fixed version would also be prudent. It will prevent those threads on the forum where people accidentally upgrade OH and it breaks things. 

Upgrading to stock Mosquitto which now supports websockets without needing to be recompiled wouldn't be a bad idea either. I believe there are CVEs for the currently installed version that I, as a security engineer, would be remiss if I didn't recommend the upgrade to fix. It should make long term support and upgrades a little easier too.

I haven't been running on an RPi 3 so I can't say what would work best for the Pi 3 or not. There is already some checking in the kiosk-xinit.sh script to do something different if it looks like it's on a more powerful machine. Perhaps it can set a variable for what load is appropriate if it is on a more powerful machine and use 2.5 for the Zero.

I'll get a PR packaged up and submitted sometime this week. I want to be very deliberate about making sure all the changes get moved over so it will take some time.

Should the openhabloader.sh script be removed? My understanding is the kiosk-xinit.sh takes it's place. Are there any other scripts in there that are no longer needed?

gulliverrr, 2020-04-15:
Can this be closed @rkoshak

rkoshak, 2020-04-16:
As far as I know, I'm the only one who has successfully installed and tested the JSONDB configs. It works for me! ;-)

But since the initial commit has been merged without that testing, I guess we can close this and open new issues as problems are found with the new configs. 

At what point can/should I start committing to docs? Is there a way to only show the officially released docs? I don't want to confuse users who are still using the old configs when they go to the docs and see something completely different. 

I will go ahead and start posting some tutorials to the forum for how to do some things through PaperUI now.

closed 2020-04-16 by rkoshak

gulliverrr, 2020-04-20:
> Upgrading to stock Mosquitto which now supports websockets

@rkoshak what do you mean? I don't see [any change](https://packages.debian.org/jessie/mosquitto).

rkoshak, 2020-04-20:
Running apt-cache madison mosquitto on the HestiaPi shows the latest available version is 1.6.7.

```
pi@spare:~ $ apt-cache madison mosquitto
 mosquitto | 1.6.7-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.6.6-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.6.4-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.6.3-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.6.2-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.6.1-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.8-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.6-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.5-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.4-0mosquitto2~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.4-0mosquitto1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.4-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5.3-0mosquitto1~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.5-0mosquitto2~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.4.15-0mosquitto4~jessie1 | https://repo.mosquitto.org/debian/ jessie/main armhf Packages
 mosquitto | 1.3.4-2+deb8u4 | http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
```

Websocket support was added to Mosquitto around version 1.5.1. https://mosquitto.org/blog/2018/08/version-151-released/

The current version of Mosquitto installed is 1.4.9

```
pi@spare:/etc/mosquitto $ mosquitto --version
Error: Unknown option '--version'.
mosquitto version 1.4.9 (build date 2019-10-30 12:02:46+0000)
```

The project provides their own apt repo if you want to keep ahead of the Debian repo which is often lagging behind, but the latest mosquitto 1.6.7 is in the apt repo.

gulliverrr, 2020-04-30:
It turned out to be pretty tricky getting the latest mosquitto run with websockets in Pi.
In fact the latest version I managed so far is 1.4.15. Any 1.5 or 1.6 cause infinite reconnections to the client (kweb) till it dies. To leave kweb out of it, I tried running the UI on my laptop while mosquitto was running on my HestiaPi and it did the same to my Chromium/FF once the first websocket connection was made. Tried a clean installation too with no compiled code. Same issues. Will keep looking for a solution...

rkoshak, 2020-05-01:
Weird. I know that the web sockets in Mosquitto 6 are being used. And from what I could tell, it's basically the same code as was compiled into version 1.4. But maybe there is something different about it that makes it hard to manage. I wonder if they add some sort of connection ACK that works differently from 1.4 so kweb isn't aware that it has successfully connected and tries again...

gulliverrr, 2020-05-01:
I tried again another fresh install of 2017-07-05-raspbian-jessie-lite skipping openhab, java and kweb and installing latest mosquitto (1.6.8-0mosquitto1~jessie1):
```
wget http://repo.mosquitto.org/debian/mosquitto-repo.gpg.key
sudo apt-key add mosquitto-repo.gpg.key
cd /etc/apt/sources.list.d/
sudo wget http://repo.mosquitto.org/debian/mosquitto-jessie.list
sudo apt-get update
sudo apt-get upgrade
mosquitto --version
sudo apt-get install mosquitto
```
Running it with websockets enabled I get this error
```
mosquitto -c /etc/mosquitto/conf.d/mosquitto.conf
1588365690: Error: Websockets support not available.
1588365690: Error found at /etc/mosquitto/conf.d/mosquitto.conf:4.
```

Downgrading to 1.6.3 starts fine:

```
sudo apt-get install mosquitto=1.6.3-0mosquitto1~jessie1
mosquitto -c /etc/mosquitto/conf.d/mosquitto.conf
1588366315: mosquitto version 1.6.3 starting
1588366315: Config loaded from /etc/mosquitto/conf.d/mosquitto.conf.
1588366315: Opening websockets listen socket on port 9001.
1588366315: Opening ipv4 listen socket on port 1883.
1588366315: Opening ipv6 listen socket on port 1883.

```
But once I connect MQTT Explorer (didn't know of this sw till you mentioned it some time ago - thanks :) ) with ws as the protocol, I get this nice CPU spike on my laptop. Even the screenshot app was affected by that load causing some ripping:
[image]

Will keep investigating...

gulliverrr, 2020-05-01:
Tried on Stretch first. Installing mosquitto (1.6.8-0mosquitto1~stretch1) installed these too:
`libev4 libuv1 libwebsockets8`

and websockets worked and connected with no problem. Will not test on Buster as I believe my hypothesis was valid. I will just investigate what is happening on Jessie...

rkoshak, 2020-05-04:
Glad to hear you are making progress and I'm sorry that it wasn't just a drop in replacement.

Though given it worked just fine on Stretch, would it make sense to spend the effort to move to Stretch rather than trying to make it work on Jessie? I know there is no problem from the OH side of things. I can't speak from the kweb side of things and I know you mentioned the bootstrap program you use to let the users enter their wifi stuff doesn't work on Buster yet.  I believe Jessie will lose LTS this June (at least Debian is dropping it, I'm not sure about Raspbian) but Stretch has support to 2022 which gives everything some time to get working on Buster or whatever will follow before LTS is lost on Stretch.

gulliverrr, 2020-05-04:
Stretch has been avoided due to the additional (total) boot time, including the UI.
With your changes I believe you have a very valid point.
I'm starting a new issue to stop hijacking this innocent one.