HestiaPi/hestia-touch-openhab #39: While loop missing if Xorg doesn't start

raycekar, 2020-04-01:
In reading through the kiosk-xinit.sh script, it seems there should be a while : that encompasses lines 22 to the end of the script.  Why I ask is because on line 167, we state Xorg is going to retry but where in the script would it retry? I see the script would terminate at that point.

[https://github.com/HestiaPi/hestia-touch-openhab/blob/ONE/home/pi/scripts/kiosk-xinit.sh#L22](url)
[https://github.com/HestiaPi/hestia-touch-openhab/blob/ONE/home/pi/scripts/kiosk-xinit.sh#L169](url)

Let me know if I am mistakenly missing something

gulliverrr, 2020-04-01:
The script does not terminate. It stays in [line 158](https://github.com/HestiaPi/hestia-touch-openhab/blob/d9594ca95c8d2880fe8de9049cb47ab44f48ff8f/home/pi/scripts/kiosk-xinit.sh#L158) as the  kweb command does not return unless it dies which will wait a bit and continue that while loop from the top again (and again and again)

raycekar, 2020-04-01:
What I am referring to is [https://github.com/HestiaPi/hestia-touch-openhab/blob/ONE/home/pi/scripts/kiosk-xinit.sh#L22](url) where if that if statement fails, it goes to [https://github.com/HestiaPi/hestia-touch-openhab/blob/ONE/home/pi/scripts/kiosk-xinit.sh#L166](url) and then echos "Xorg not running. Retrying...", sleeps, and then the scripts stops.  At this point I am asking where it would retry at?  I there something up the stack that would recall this script to have it retry?

gulliverrr, 2020-04-02:
Apologies for missing your point earlier. 
You are absolutely right. Feel free to submit a PR or I will do it when I get the chance.
For your reference the boot sequence is described [here](https://github.com/HestiaPi/hestia-touch-openhab/wiki/Boot-sequence-ONE).
Well spotted! The joy of open source ;)

commit b58a32c8 (2020-04-18): Retry if Xorg doesn't start 

Fixes #39
closed 2020-04-18 by gulliverrr