HestiaPi/hestia-touch-openhab #41: String item value shown as text next to selection

gulliverrr, 2020-04-17:
Basic UI shows ON/OFF/Boost item value as text next to selection buttons:
[image]
I find it redundant as selected button shows status.
Removing `[%s]` from Paper UI>Configuration>Items did not affect Basic UI.
This was caused after #38 @rkoshak

rkoshak, 2020-04-17:
That's an easy fix. But removing the `[%s]` isn't enough. This is a long running bug. But you can suppress it using `[]` instead, though I ended up fixing it in the sitemap. Where do you perfer I apply the fix? The sitemap or on the Item?

gulliverrr, 2020-04-17:
I assume the Item is where is should originate correctly although I understand it is a presentation issue.
This is happening for all selection items too (Temperature Unit,  System Type and 2nd Stage Heating)

rkoshak, 2020-04-17:
After looking at it, I think I want to make the fix on the sitemap instead of on the Item itself. If I make the change to the Item, it breaks the Control tab of PaperUI so you no longer see the state of the mode there.

gulliverrr, 2020-04-22:
Fixed by #43

gulliverrr, 2020-04-22:
Fixed by #43

closed 2020-04-22 by gulliverrr