HestiaPi/hestia-touch-openhab #42: Current Temperature on LCD has 3 decimal points for Celsius mode

gulliverrr, 2020-04-17:
In Celsius mode, current Temp on LCD has 3 decimal points.
Value on MQTT topic is actually published with 3 decimal points (22) but format `Temperature Setpoint [%.1f °C] ` is not respected. Is this related to the bug suggested for #41 ?
Basic UI has no issue.
This was caused after #38 @rkoshak

rkoshak, 2020-04-17:
The format only applies to the label as it is displayed on the sitemap so it's not related to #41. I'll look more into it. This part is basically unchanged from the original config but there may have been some minor changes to the MQTT binding itself that this is related to. I think I can sue the outgoing format to reduce it to one decimal place. If not, I can use an outgoing JS transformation.

rkoshak, 2020-04-22:
Found a solution for this. If you don't have any other transformations on the Channel, there is an "Outgoing Format" field where you can apply the %.1f to round it to one decimal place before publishing it. Thankfully it doesn't appear to add a decimal place for F so this should fix it. 

I'm liking how little the changes are to fix these problems. Expect a PR shortly.

gulliverrr, 2020-04-22:
Fixed by #43

closed 2020-04-22 by gulliverrr