HestiaPi/hestia-touch-openhab #47: Add weather info

rkoshak, 2020-04-29:
This is definitely an enhancement and much lower priority than other enhancements but I wanted to capture it here.

OH supports a number of weather services with bindings or one can access them through the HTTP binding. We could also set up an MQTT topic that users could publish current outside temp and conditions to. It would be kind of nice to see the outside temp and a little current conditions icon on the LCD. Knowing the outside temp can be useful for some users who want to create custom rules that depend on the outside temp.

If I knew how to update the LCD UI, maybe this is better left as a tutorial on the forum or in the docs?

It would require some configuration by the end user to set up correctly, mainly setting their location in PaperUI and filling in their API information to access the weather service, though I think OH automatically discovers the the location based on the external IP address. Regardless, the UI would have to be smart enough to hide the weather info when there is no info to show.

Thoughts?

gulliverrr, 2020-04-30:
As the UI is already setup around MQTT and websockets, it would be very easy to add some icon(s) and text to it as long as it is published on a topic. The ONE UI is entirely @jaythomas work and Vue is something new to me but if Jay is busy I can give it a go although I am not the person to decide what appears where on the LCD.

jaythomas, 2020-05-03:
There is very little screen real estate to work with, so I'm not sure what you had in mind on how to display it and how verbose the information should be.

I suppose if the user tapped the house temperature the :house: icon could change to something representing outside... like a cloud, sun, thunder cloud, etc. and display the weather API temperature instead, (and display the outside humidity in place of the indoor humidity.) This wouldn't be intuitive to the user though and could even lead to confusion as to whether they are display the indoor or outdoor temperature.

You could also create a separate page accessible from a tap in the top-right corner similar to how you have the :information_source: icon in the top-right corner now that displays system hardware information, but having 2 icons in the top-right corner is pushing the limit on what you can do with the limited space.

Any other ideas I can think of would require reworking the placement of everything and a lot of work.

rkoshak, 2020-05-04:
At most the temp, the humidity, and an icon Even just the outside temp and icon would probably be sufficient for my thoughts. I was thinking that there could be another row below the sensor readings showing the outside temp and icon or outside temp, humidity, and icon. The sensor readings would probably need to be shrunk a little bit to make room.

If it will be too much work I can understand that. I'm not in a position where I can help so it's not fair to impose on others. I might experiment with my own system to see if I can make it work or switch back over to HABPanel or something like that.

atd, 2020-06-08:
In case it gives any idea, this is how I am showing it in my kitchen computer screensaver (with touchscreen)

Of course, there is plenty much more space than in the LCD

[image]