HestiaPi/hestia-touch-openhab #51: Move to Raspbian Stretch

gulliverrr, 2020-05-04:
With @rkoshak greatest and latest changes #37 , boot time has been dropped to 10 minutes. This is the best excuse to start working towards a Stretch based HestiaPi ONE build.

rkoshak, 2020-05-04:
Here's a question, assuming we get stock Mosquitto with websocket support working well, is Turnkey the only thing keeping us from upgrading to Buster now, or is there anything else? I did some looking and it very much looks like Turnkey is abandoned (last commit was over two years ago, last open PR has been open for just about that long) which means that at some point it will have to be replaced or HestiaPi will need to take on maintaining their own fork of it or be stuck on an unsupported base OS in two years (assuming this Issue get's addressed properly).

At first glance it doesn't appear like there is anything that would make it not work on Buster, but I plan on testing it out this week sometime just to verify (and use in some of my own projects ;-) ). Do you do all the optional steps for HestiaPi (install Node.js, install go, etc.)? I don't need it for my stuff but I may as well test it out in a way that works for HestiaPi at the same time.

gulliverrr, 2020-05-06:
> Do you do all the optional steps for HestiaPi (install Node.js, install go, etc.)?

No optional steps are followed. 

I will try Stretch first as I have most of the steps documented (but not published) and we take it from there.

raycekar, 2020-10-16:
Now I just did a quick google search on Turnkey and see that @rkoshak is right it hasn't been updated for quite a while. Now I know I don't have all of the technical details, but as an idea, what would you guys think of using the touch screen where on boot, it has an onscreen keyboard first asking for network name, then network password?  Eliminating Turnkey and then being able to go to Buster.

gulliverrr, 2020-10-16:
The big issue we faced trying to move to Stretch or Buster is the LCD drivers together with the fact that we need a Lite setup without Desktop and our browser in Kiosk mode. Turnkey is not much more than "a script" that turns on and off native Pi's services. I don't think it will be an issue we cannot solve once we solve the rest.

hestiahacker, 2022-10-22:
Progress has been made on this. See https://community.hestiapi.com/t/automated-build-process-stretch/1965 for details.

The TL;DR is that there's now a stretch image but we're still using the same old versions of everything for now. If this works as expected, we will know we have a solid base and can try switching over to the repo versions of things one at a time to see which of the stock packages will work versus what will not.

hestiahacker, 2023-12-07:
This ticket can be closed. We moved from Stretch to Buster to Bullseye.