HestiaPi/hestia-touch-openhab #53: IP info displays error when no wi-fi

jaythomas, 2020-06-03:
Expected:
When there is no wi-fi, display nothing (empty string)

Actual results:
A curl error is displayed, pushing the rest of the system info off the LCD screen
[image]

gulliverrr, 2020-06-03:
I believe this is sorted by [this](https://github.com/HestiaPi/hestia-touch-openhab/commit/69d1428b4b38c4cc750c6ffee366ea6cde55dcb9#diff-4e46f298b1efea409f906671bd5067bc) commit. Could you please check the contents of your `home/pi/scripts/getpublicip.sh` if it has this extra `while` check on the top?

jaythomas, 2020-06-04:
I believe this does the trick, but if not I will re-open if I see the issue again. Thanks.

closed 2020-06-04 by jaythomas