HestiaPi/hestia-touch-openhab #56: Missing close quote in rule TempUnit changed

gulliverrr, 2020-07-28:
@rkoshak creating the new image file stumbled upon this one:
In rule TempUnit changed there is a closing quote at the end. Please change this line as I don't how to produce the commit for the new rules

    var comfDef = DEFAULTS.get(newState+"_"+comfMode+"_DEF); 
to this:

    var comfDef = DEFAULTS.get(newState+"_"+comfMode+"_DEF");

rkoshak, 2020-07-28:
I don't want to be the single point of failure on how to update the rule. The steps to update a rule and get it committed is as follows:

1. Make the changes on the running openHAB instance.
2. Test the changes.
3. The changed rules are located in /var/lib/openhab2/jsondb/automation_rules.json. Copy that file to hestia-touch-openhab/var/lib/openhab2/jsondb.
4. Add, commit and push and it's done.

The humidity rule error is almost certainly because HumidityRemBoostTime is NULL. I'll add a check.

The last error is a typo. It should be `events.sendCommand`, not `events.sendcommand`. It doesn't provide which script it occurred in so the way I found it was to open the automation.json file with `less` and searched for it. Turns out it's in the "SystemType changed".

gulliverrr, 2020-07-28:
Thanks for the fix and the detailed steps on where to find the file to commit back here.

commit 8a9e415b (2020-07-28): Merge pull request #57 from rkoshak/minor-bugs

Fixed some minor bugs, all are typoes. Fixes #56
closed 2020-07-28 by gulliverrr