HestiaPi/hestia-touch-openhab #61: No longer receiving temperature MQTT event on subscribe

jaythomas, 2020-10-11:
The `hestia/local/temperature` event only comes across when the temperature changes, which means the UI will display "0" degrees indefinitely in certain situations, such as bringing up the UI after OpenHab2 is running, subscribing from an external device, or serving up the Vue frontend on the local network for development purposes.
I haven't noticed this bug in the past so I think it may be a regression since the v1.2 refactoring.

gulliverrr, 2020-10-22:
I believe this is only true when openHAB first starts. Once it starts and update is received (temperature changes) then the last value is "retain"ed on that topic so whenever afterwards someone subscribes to that topic, they will receive the latest value. @rkoshak am I wrong?

jaythomas, 2020-10-22:
To replicate it, try running the frontend locally and giving it the ip of an already-running thermostat (`MQTT_SERVER=192.168.1.23 yarn serve`) and when you open the network tab you will see every request come through for initial parameters **except** temperature. The UI will display "0". Wait so many minutes until the temperature changes and you will receive the update, but like I mentioned it could display "0" for a very long time if the temperature does not fluctuate.

rkoshak, 2020-10-27:
Jay's right. The retained flag isn't set. Given how ephemeral the temperature readings are I'm not sure the solution is to set the retained flag. If something goes wrong that temperature reading can become very old and there's no way to tell. This is unfortunate as the solution would be to just toggle the retained flag.

So, if we want to publish the updates even when they are the same, we need to change all the triggers in the "Synchronize Temp Proxies" rule to use received update instead of changed. 

It'll take me a few days to get to this but it is not hard if either of you want to take a shot. All you need to do is create a new rule trigger for each of the Items using "received update" instead of "changed" and then delete the "changed" triggers.

Alternatively, stop openHAB and edit /var/lib/openhab2/jsondb/automation_rules.json and change the trigger types from `core.ItemStateChangeTrigger` to `core.ItemStateUpdateTrigger` for the rule triggers on that rule. Search for "Sync" to find the "Synchronize Temp Proxies" rule and the rule triggers will be in the JSON section above that.

All of the Temp proxy Items should change but I don't know about the setpoint Item triggers. Perhaps they can be kept as is.

jaythomas, 2020-10-27:
My hands are tied for the next 3 weeks unfortunately. I wouldn't say this is any rush either though. Outside of development it only becomes an issue when you want to set up a remote control for the thermostat (a touchscreen in another room running the ui). I haven't seen any other community members trying this yet.