HestiaPi/hestia-touch-openhab #89: Memory exhaustion causing severe touch interface lag on Pi Zero 2 W - Configuration fix included

AdamGoyer, 2025-09-18:
Hello HestiaPi community,

I'm a new user experiencing significant performance issues with my HestiaPi Touch system. After extensive troubleshooting, I've identified the root cause and have a working solution that might help others. I wanted to share my findings and see if this could potentially improve the default configuration.

### Environment
- **Hardware:** Raspberry Pi Zero 2 W (quad-core ARM Cortex-A53)
- **HestiaPi Version:** v1.4-dev (OpenHAB 2.5.3)
- **Installation Date:** Recently installed by integrator
- **Symptoms:** Touch interface requiring 2-3 second press-and-hold, severe lag, nearly unusable

### Problem Description

The touch interface was essentially non-functional with symptoms including:
- Excessive pressure required for touch registration
- 2-3 second delays for any UI response
- System load average of 2+ (on the 4-core Pi Zero 2 W)
- Swap usage at 98-100% constantly
- Touch events queuing up and executing minutes later

Initially suspected hardware issues (loose cables, touchscreen calibration) but diagnostics showed the hardware was functioning correctly.

### Root Cause Analysis

After SSH access and system monitoring, discovered:
```bash
# Memory status before fix:
Memory: 273MB used / 427MB total (64%)
Swap: 99MB / 99MB (100% FULL)
OpenHAB Process: 215MB (49.3% of total memory)

# Java configuration check:
grep EXTRA_JAVA_OPTS /etc/default/openhab2
EXTRA_JAVA_OPTS=""  # No heap limits!
```

The issue: OpenHAB's Java process was running without heap constraints, allowing it to consume >50% of available RAM and forcing constant swap thrashing.

### Solution Applied

Added Java heap limits optimized for Pi Zero 2 W's limited RAM:

```bash
# Backup original configuration
sudo cp /etc/default/openhab2 /etc/default/openhab2.backup

# Edit /etc/default/openhab2
EXTRA_JAVA_OPTS="-Xms96m -Xmx192m -XX:ParallelGCThreads=2"

# Restart OpenHAB
sudo systemctl restart openhab2
```

### Initial Results

After applying heap limits and restart:
- **Swap usage:** Dropped from 98% to 63% (35MB freed)
- **Touch response:** Noticeably improved, though still testing
- **System load:** Currently ~2.2 on 4 cores
- **Memory pressure:** Reduced but needs longer-term monitoring

**Note:** These settings were just applied, so long-term stability is still being evaluated. Will update with more data after extended testing.

### Potential Considerations

1. **For Pi Zero 2 W users:** The 192MB heap limit might be a good starting point, though optimal values may vary
2. **Default configuration:** Heap limits could potentially help on memory-constrained devices
3. **Further testing needed:** To determine if these settings are sustainable over days/weeks

### Questions for the Community

1. Are there any downsides to these heap limits I should be aware of?
2. Has anyone found a better configuration for Pi Zero 2 W?
3. Is there a reason the default configuration doesn't include heap limits?
4. What heap settings are others using successfully on similar hardware?
5. Should I be concerned about the G1GC warning (seems to fall back to parallel GC)?

I hope this helps others experiencing similar performance issues. Happy to provide more details or test alternative configurations if needed.

Thank you for this great open-source project!

---

*Note: I'm new to the HestiaPi community, so please let me know if I've missed any important context or if there are better ways to address this issue.*

gulliverrr, 2025-09-18:
Have you tried a different SD card? Your symptoms are not reproducible. The [forum](https://community.hestiapi.com/) is the best place to take this conversation instead of GitHub

closed 2025-09-19 by AdamGoyer

gulliverrr, 2025-09-19:
There have been many times where users face similar symptoms and are solved by replacing the SD card with a branded higher speed so we would suggest redownloading the image and reflashing it to a new card.